KaeptnBalu/Arduino_Spot_Welder_V4 #1: LiFePO4 Batteries

DanielMarcato, 2021-05-17:
Hi,

no other idea how to ask this question so I will try to get an answer via an "issue".

Do you think the new V4 can handle car batteries made from LiFePO4 cells?
I have access to one made of four 280Ah cells in series which gives a ~13.4V total.
Current rating from the manufacturer is 280A continuously and I don't have any information on pulse capabilities but my guess is that the lithium version could be much more powerful than an average car battery.

Looking forward to your response!

Best,
Daniel

KaeptnBalu, 2021-05-18:
Hi,
I have no acess to such a lithium car battery, so dont know how much max. current it can put out.
The welder can handle up to 800A no problem. Maybe even up to 1000A could be fine.
Also you need to check if your lifepo4 battery is protected by a bms. Then it is possible that your battery is shut down after every weld because of over current / short circuit protection.

DanielMarcato, 2021-05-18:
BMS is installed but I put the V1 welder directly to the battery terminals thus circumventing the BMS. The V1 died (first the TVS diodes that I had installed and then one of the FET driver chips) which is why I ordered the V4 now.

Maybe a list in your repositories with successful battery setups would be useful - people could post their batteries and weld times to get a feeling for good brands etc.

DanielMarcato, 2021-05-23:
Worked perfectly with the Lifepo - great product!

[image]

closed 2021-05-23 by DanielMarcato