KaeptnBalu/Arduino_Spot_Welder_V4 #3: Some corrections in excel BOM and mouser basket

Nick-Z0, 2023-01-29:
Hey Marc, first of all thanks for this great product!

I have found a couple of issues in the provided BOM excel file listed below:

**Excel BOM**
Line 	problem
6		mosfet quantity 6 instead of 8 (for new version)
17		1 schotky instead of 2
16		IC socket not needed
22		film capacitor 1μF RM5
20		TVS diode (updated in new version?)
24		capacitor not in stock (I used this one as a replacement: 661-EMZL250ARA471MJA)

**Mouser basket issues**
3.3k Ohm resistor wrong quantity 10 instead of 1 (?)
Maybe also add optional 12V 30x30mm fan link

I also have a question. Will a thicker cable be any improvement for the welder?

KaeptnBalu, 2023-01-31:
Thanks for pointing this out. I fixed the errors in the BOM.
The capacitor you chose as replacement for the out of stock one should work fine.

Mouser basket issues:
There is just 1 resistor 3.3k Ohm necessary. Anyway i changed the quantity to 10 in the basket so now all resistors are listed with quantity 10 to have spare.

Thicker cable can be used if you need longer than the standard 50cm welding leads. For the standard 50cm it is not necessary to go with thicker cables.

Nick-Z0, 2023-01-31:
Do you have a schematic or instructions for the PCB assembly? Or are the BOM descriptions (excel file) sufficient?

Also, I could not edit the mouser basket because I already had some parts. I had to share it to myself and then I could edit it.

KaeptnBalu, 2023-02-01:
There is no assembly instructions for the pcbs.
But you can find the schematic and pcb layout here: https://github.com/KaeptnBalu/Arduino_Spot_Welder_V4/tree/main/KiCAD_PCB_Files
The "PCB Reference" mentioned in the BOM is printed on the pcbs for every part, so it is pretty easy to populate the pcbs.

closed 2024-07-03 by KaeptnBalu