KaeptnBalu/Arduino_Spot_Welder_V4 #5: Using Cap Bank

BenJBachler, 2024-09-22:
I would like mod this use a capacitor bank as a current source and a lab bench power supply as a power source instead of a battery. I'm thinking of adding a 3rd board to mount the capacitors and a mosfet to disconnect the power supply once charged. Is there an engineering reason this wasn't done before and would it be possible to integrate this into your design?

KaeptnBalu, 2024-09-23:
It was not done yet because the necessary capacitor bank to deliver enough power is much more expensive than a car battery or lipo. There is no technical reason against this idea.

BenJBachler, 2024-09-24:
This is more in depth than I thought, if you don't mind helping with this. The battery pack would provide constant voltage and the max current that the welding surface will allow while a CD welder's voltage would drop as the charge is drained. How much does this affect the weld? I am thinking about taking the route shown [here](https://www.zeva.com.au/Projects/SpotWelderV2/) and replace your Mosfet board. I think I would have to tailor your code to make this work though. Another idea that would be simpler would be to use a dewalt battery and a regulator board to provide the VCC voltage for the nano but I don't think the mosfet can handle the extra voltage (dewalt is 20V). Would using a dewalt battery be too much voltage?

closed 2024-10-14 by BenJBachler